khanning/scratch-arduino-extension #12: Computer reboots when using scratchx arduino experimental interface

emeyedeejay, 2015-05-13:
I followed the steps to enable communication between my Arduinoi and ScratchX using the experimental Arduino extension.

I then experienced 4 reboots of my PC once I had started using the interface. Seemed to be at random times into working with the interface but all within a minute or 2. I was just experimenting with blinking a LED - nothing fancy.

This seems directly realated to using the scratchx interface. It started directly after configuring it all up (enabling chrome add-on etc.) and only happens once I drag an Arduino related block or 2 to the interface.

Adruino Nano
Windows 8.1
Chrome (unknown version right now)

damellis, 2015-05-13:
Weird. Does it happen if you launch ScratchX without having an Arduino board connected? Have you used Arduino on the computer before? Ever gotten reboots?

emeyedeejay, 2015-05-13:
-I'll try it with out the arduino attached tonight.
-Used the Arduino plenty before
-No random reboots at all.

The timing coincided exactly with getting scratch going - I got about 3
blinks during the longest session so that was probably at most 1.5 minutes
in ... just dragging the When device is connected, connect LED A to pin 6,
turn on LED A, pause 1 sec, Turn off LED A, Pause 1 sec. repeat ... 2/3
blinks and boom - crash :(

emeyedeejay, 2015-05-24:
I went to test this again this morning (been travelling for work so have
had zero time!) but I now can;t launch the intervface.

I just continually get an "Aw Snap = Something went wrong while displaying
this web page message". Tried a couple of things but can;t get past this
stage.

Is this a known issue at all?

Thanks!

emeyedeejay, 2015-05-24:
Did some troubleshooting ... uninstalled the plugin, rebooted.

Scratchx interface with Arduino blocks loaded fine.
Re-downloaded the plugin
Exited chrome
Confirmed NPAPI is enabled
Installed plugin
Relaunched Chrome
Browsed to scratchx
Clicked "arduin"
Same error - "Aw Snap" error message/

I'll fire up another bug report? 
Testing with IE now [shudder]

emeyedeejay, 2015-05-24:
Happens in incognito mode as well so doesn't look like an extension conflict.

emeyedeejay, 2015-05-26:
Any ideas at all anybody? I can't get further than the "aw snap" error page. Thanks!

damellis, 2015-05-26:
Hmm, Google has some suggestions here: https://support.google.com/chrome/answer/95669?hl=en

Does any of that apply / help?

emeyedeejay, 2015-05-26:
Thanks David.

I've been through them....

I tried on my [work] laptop as well and I get exactly the same result so I suspect on of my chrome extensions (becuase they sync across both my home PC and work laptop).

I disabled all my extensions in incognito mode and tried that too - same error :sigh:

I'll keep troubleshooting (not sure what's next).

khanning, 2015-05-26:
I just tested this on a Windows 8.1 VM and I'm getting the same result you are. It looks like it's a larger issue with ScratchX/Scratch Browser Plugin.

I filed an issue under the llk/scratchx repo
https://github.com/LLK/scratchx/issues/30

emeyedeejay, 2015-05-26:
Thanks Kreg ... glad it's not just me!

If I can test on that issue let me know - it's very odd because I got it running previously but my PC rebooted whenever I started using it (which is what this issue is about :)

emeyedeejay, 2015-06-08:
Does anybody have any idea on how to troubleshoot/resolve this? It's so frustrating :)

khanning, 2015-06-17:
@emeyedeejay the issue with the browser plugin on Win 8 has not been addressed yet. Keep an eye on [this issue](https://github.com/LLK/scratchx/issues/30) for any updates. I will also post here when the issue is resolved.