khanning/scratch-arduino-extension #14: Order of Blocks

cymplecy, 2015-05-15:
As discussed by email
A lot of the blocks are Grove specific.

I'd suggest moving them down the list and promoting generic Arduino pin writing/reading blocks to the top

damellis, 2015-05-22:
The blocks aren't intended to be Grove-specific, just sensor or actuator specific. That is, the servo blocks will work with standard servos, not just the grove ones; same with the other components. We show the Grove modules in the documentation because they're easy to get started with, but the blocks will also work with basic electronic components on a breadboard. (We're hoping to document the relevant circuits, soon.)

We put the component-specific blocks at the top because they seemed to be a more natural fit with the level of abstraction found elsewhere in Scratch. We liked the idea of working with component-specific language (e.g. "button pressed") as a starting point. The generic Arduino blocks felt more like an advanced / low-level interface that people might progress too when they decide they need more flexibility than they get with the component specific blocks.

Does that make sense?

cymplecy, 2015-05-22:
I must admit that I didn't look too deeply at what the blocks at the top actually did ( I just assumed LEDA was a Grove thing) and I now see what you've done.  Apolgies.

I see your philosophy but I think its not worth the confusion for beginners.

Its an easy concept to connect (say a LED) to a pin and then say Pin on

But having to map a name, say LEDA, to a physical pin and then turn it on is an unnecessary step for a beginner :)

I work with primary age (7-11) year olds in  my day job and I'm a passionate believer in one of the main Scratch concepts - low barrier to entry

I can see that you've attempted to do that but I think you've just swapped one barrier for one a little bit higher (not much but higher)

Simon
PS I'm the author of ScratchGPIO for the Pi and I've been developing Scratch syntax for pupils for the past 3 years :)

khanning, 2015-05-22:
Simon, thank you for all your feedback! Lowering the barriers to entry and reducing complexity is definitely a top priority for this project. Maybe this example will help make the case for component-specific blocks. 

Reading a value from a rotation knob (potentiometer) and using that value to adjust an LED's brightness could be done one of the following ways:

[image]

-or-

[image]

While the second example uses less blocks, I would argue that the first example is easier to understand by just reading it. 

Defining the pin assignments at the beginning of the script also makes it much easier for someone else to wire up their Arduino to run your project.

cymplecy, 2015-05-22:
You make a good case :)

I agree that once people have stepped over your higher step :) then life is easier.

But you have added a step :)

Lets see if we can come to some compromise

At a minimum - change the word connect to something else - not sure what at moment but we are not connecting anything- we are mapping but I don't think we should use that word either.

taking 
set [Led A] brightness to read [rotation knob] %
isn't good syntax either :)

1st suggestion would be

set [ledA] power to reading of [rotation knob] 

I've spent a long time considering best way of describing level/brightness/speed of motors and concluded that power is best compromise between physics and language

Technically its not power but its easy for kids to understand that more power to LED == more light, more power to motor == more speed

But moving onto plain usage change
set pin 3 to read analog 0 %

to
set pin [3] power to analog input [0]

or

set pin [3] power to sensor input [0]

just as in plain Scratch with Picoboard and a variable called test you'd say
set [test] to sensor-value [slider]

Simon

damellis, 2015-05-22:
I'd echo what Kreg said. On the other hand, I haven't tried the extension with kids yet, so it may be that we're wrong in our assumptions. It would be great to get feedback as you or others try it out.

cymplecy, 2015-05-22:
em - its doesn't actually work for me on my win 8.1 machine!  But I'll let others work on that aspect :)