khanning/scratch-arduino-extension #15: Language translations for Scratch blocks

khanning, 2015-05-21:
If anyone is interesting in translating the Scratch blocks into another language please submit a pull request or just comment on this issue with your translation.

To submit a pull request:
--
1. Create a copy of the arduino_extension.js file and append the two or three digit [ISO 639 Language Code](http://www.loc.gov/standards/iso639-2/php/code_list.php). 
e.g.
arduino_extension_es.js (Spanish)
arduno_extension_de.js (German)

2. Translate just the bolded text in the block descriptor:
<pre><code>var descriptor = { 
    blocks: [
      ['h', '<b>when device is connected</b>', 'whenConnected'],
      [' ', '<b>connect %m.hwOut to pin %n</b>', 'connectHW', 'led A', 3], 
      [' ', '<b>connect %m.hwIn to analog %n</b>', 'connectHW', 'rotation knob', 0], 
      ['-'],
      [' ', '<b>set %m.leds %m.outputs</b>', 'digitalLED', 'led A', 'on'],
      [' ', '<b>set %m.leds brightness to %n%</b>', 'setLED', 'led A', 100],
      [' ', '<b>change %m.leds brightness by %n%</b>', 'changeLED', 'led A', 20],
      ['-'],
      [' ', '<b>rotate %m.servos to %n degrees</b>', 'rotateServo', 'servo A', 180],
      [' ', '<b>rotate %m.servos by %n degrees</b>', 'changeServo', 'servo A', 20],
      ['-'],
      ['h', '<b>when %m.buttons is %m.btnStates</b>', 'whenButton', 'button A', 'pressed'],
      ['b', '<b>%m.buttons pressed?</b>', 'isButtonPressed', 'button A'],
      ['-'],
      ['h', '<b>when %m.hwIn %m.ops %n%</b>', 'whenInput', 'rotation knob', '&gt;', 50],
      ['r', '<b>read %m.hwIn</b>', 'readInput', 'rotation knob'],
      ['-'],
      [' ', '<b>set pin %n %m.outputs</b>', 'digitalWrite', 1, 'on'],
      [' ', '<b>set pin %n to %n%</b>', 'analogWrite', 3, 100],
      ['-'],
      ['h', '<b>when pin %n is %m.outputs</b>', 'whenDigitalRead', 1, 'on'],
      ['b', '<b>pin %n on?</b>', 'digitalRead', 1], 
      ['-'],
      ['h', '<b>when analog %n %m.ops %n%</b>', 'whenAnalogRead', 1, '&gt;', 50],
      ['r', '<b>read analog %n</b>', 'analogRead', 0], 
      ['-'],
      ['r', '<b>map %n from %n %n to %n %n</b>', 'mapValues', 50, 0, 100, -240, 240]
    ],
</code></pre>

3. Submit a pull request to add the newly translated extension to the root of the master repo.

Thank you!

christianimmler, 2015-05-21:
This is a german translation:

var descriptor = { 
blocks: [
  ['h', 'Wenn Arduino verbunden ist', 'whenConnected'],
  [' ', 'Verbinde %m.hwOut mit Pin %n', 'connectHW', 'led A', 3], 
  [' ', 'Verbinde %m.hwIn mit Analog %n', 'connectHW', 'rotation knob', 0], 
  ['-'],
  [' ', 'Schalte %m.leds %m.outputs', 'digitalLED', 'led A', 'on'],
  [' ', 'Setze %m.leds Helligkeit auf %n%', 'setLED', 'led A', 100],
  [' ', 'Ändere %m.leds Helligkeit um %n%', 'changeLED', 'led A', 20],
  ['-'],
  [' ', 'Drehe %m.servos auf %n Grad', 'rotateServo', 'servo A', 180],
  [' ', 'Drehe %m.servos um %n Grad', 'changeServo', 'servo A', 20],
  ['-'],
  ['h', 'Wenn %m.buttons ist %m.btnStates', 'whenButton', 'button A', 'pressed'],
  ['b', '%m.buttons gedrückt?', 'isButtonPressed', 'button A'],
  ['-'],
  ['h', 'Wenn %m.hwIn %m.ops %n%', 'whenInput', 'rotation knob', '>', 50],
  ['r', 'Wert von %m.hwIn', 'readInput', 'rotation knob'],
  ['-'],
  [' ', 'Schalte Pin %n %m.outputs', 'digitalWrite', 1, 'on'],
  [' ', 'Setze Pin %n auf %n%', 'analogWrite', 3, 100],
  ['-'],
  ['h', 'Wenn Pin %n ist %m.outputs', 'whenDigitalRead', 1, 'on'],
  ['b', 'Pin %n ein?', 'digitalRead', 1], 
  ['-'],
  ['h', 'Wenn Analog %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
  ['r', 'Wert von Analog %n', 'analogRead', 0], 
  ['-'],
  ['r', 'Setze %n von %n %n auf %n %n', 'mapValues', 50, 0, 100, -240, 240]
],

khanning, 2015-05-22:
@christianimmler thank you for the translation! I forgot to include the drop down menus in my initial comment. I've updated the first comment to include all the text that needs translation if you wouldn't mind translate the menu wording as well.

christianimmler, 2015-05-23:
Ok, here the missing part of the translation:

menus: {
  buttons: ['Taste A', 'Taste B', 'Taste C', 'Taste D'],
  btnStates: ['gedrückt', 'losgelassen'],
  hwIn: ['Drehknopf', 'Lichtsensor', 'Temperatursensor'],
  hwOut: ['LED A', 'LED B', 'LED C', 'LED D', 'Taste A', 'Taste B', 'Taste C', 'Taste D', 'Servo A', 'Servo B', 'Servo C', 'Servo D'],
  leds: ['LED A', 'LED B', 'LED C', 'LED D'],
  outputs: ['Ein', 'Aus'],
  ops: ['>', '=', '<'],
  servos: ['Servo A', 'Servo B', 'Servo C', 'Servo D']
},

khanning, 2015-05-26:
@christianimmler thank you for the translation! You can now load the blocks in German by adding lang=de to the ScratchX project URL.

e.g.
http://scratchx.org/?url=http://khanning.github.io/scratch-arduino-extension/arduino_extension.js&lang=de

consolacion, 2015-05-26:
Dutch!. Mind you though that the correctness depends on what the actual sentence in english is supposed to look like. It is hard to properly translate a few words here and there if  one doesn't know what the full sentence would be or what the context is.

var blocks = { 
  en: [
    ['h', 'als het apparaat verbonden is', 'whenConnected'],
    [' ', 'verbindt %m.hwOut met pin %n', 'connectHW', 'led A', 3], 
    [' ', 'verbindt %m.hwIn met analoog %n', 'connectHW', 'draaiknop', 0], 
    ['-'],
    [' ', 'schakel %m.leds %m.outputs', 'digitalLED', 'led A', 'on'],
    [' ', 'schakel %m.leds helderheid tot %n%', 'setLED', 'led A', 100],
    [' ', 'verander %m.leds helderheid met %n%', 'changeLED', 'led A', 20],
    ['-'],
    [' ', 'draai %m.servos tot %n graden', 'rotateServo', 'servo A', 180],
    [' ', 'draai %m.servos met %n graden', 'changeServo', 'servo A', 20],
    ['-'],
    ['h', 'wanneer %m.buttons is %m.btnStates', 'whenButton', 'knop A', 'in gedrukt'],
    ['b', '%m.knoppen in gedrukt?', 'isButtonPressed', 'knoppen A'],
    ['-'],
    ['h', 'wanneer%m.hwIn %m.ops %n%', 'whenInput', 'draaiknop', '>', 50],
    ['r', 'read %m.hwIn', 'readInput', 'draaiknop'],
    ['-'],
    [' ', 'schakel pin %n %m.outputs', 'digitalWrite', 1, 'on'],
    [' ', 'schakel pin %n tot %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', 'wanneer pin %n is %m.outputs', 'whenDigitalRead', 1, 'on'],
    ['b', 'pin %n aan?', 'digitalRead', 1], 
    ['-'],
    ['h', 'wanneer analoge %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', 'lees analoge %n', 'analogRead', 0], 
    ['-'],
    ['r', 'zet %n van %n %n tot %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    en: {
      buttons: ['knop A', 'knop B', 'knop C', 'knop D'],
      btnStates: ['ingedrukt', 'losgelaten'],
      hwIn: ['draaiknop', 'licht sensor', 'temperatuur sensor'],
      hwOut: ['led A', 'led B', 'led C', 'led D', 'knop A', 'knop B', 'knop C', 'knop D', 'servo A', 'servo B', 'servo C', 'servo D'],
      leds: ['led A', 'led B', 'led C', 'led D'],
      outputs: ['aan', 'uit'],
      ops: ['>', '=', '<'],
      servos: ['servo A', 'servo B', 'servo C', 'servo D']
    }
};

khanning, 2015-05-26:
@consolacion thank you for the translation! You can load the Dutch blocks by adding lang=nl to the ScratchX URL.

e.g.
http://scratchx.org/?url=http://khanning.github.io/scratch-arduino-extension/arduino_extension.js&lang=nl

If any of the wording doesn't seem right once you start using the extension please feel free to submit a correction!

wilsonjuniorr, 2015-07-06:
Portuguese
var blocks = { 
  pt: [
    ['h', 'Quando dispositivo estiver conectado', 'whenConnected'],
    [' ', 'conectar%m.hwOut para pino %n', 'connectHW', 'led A', 3], 
    [' ', 'conectar %m.hwIn para analogico %n', 'connectHW', 'potenciometro', 0], 
    ['-'],
    [' ', 'estado %m.leds %m.outputs', 'digitalLED', 'led A', 'ligado'],
    [' ', 'estado %m.leds brilho to %n%', 'setLED', 'led A', 100],
    [' ', 'mudar %m.leds brilho em %n%', 'changeLED', 'led A', 20],
    ['-'],
    [' ', 'girar %m.servos para %n graus', 'rotateServo', 'servo A', 180],
    [' ', 'girar %m.servos em %n graus', 'changeServo', 'servo A', 20],
    ['-'],
    ['h', 'quando %m.buttons is %m.btnStates', 'whenButton', 'botao A', 'pressionado'],
    ['b', '%m.buttons pressionado?', 'isButtonPressed', 'botao A'],
    ['-'],
    ['h', 'quando %m.hwIn %m.ops %n%', 'whenInput', 'potenciometro', '>', 50],
    ['r', 'read %m.hwIn', 'readInput', 'potenciometro'],
    ['-'],
    [' ', 'estado digital pino %n %m.outputs', 'digitalWrite', 1, 'ligado'],
    [' ', 'estado analogico pino %n to %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', 'quando pino %n is %m.outputs', 'whenDigitalRead', 1, 'ligado'],
    ['b', 'pino %n ligado?', 'digitalRead', 1], 
    ['-'],
    ['h', 'quando valor analogico %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', 'ler valor analogico %n', 'analogRead', 0], 
    ['-'],
    ['r', 'mapear %n from %n %n to %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    pt: {
      buttons: ['botao A', 'botao B', 'botao C', 'botao D'],
      btnStates: ['pressionado', 'solto'],
      hwIn: ['potenciometro', 'sensor de luz', 'sensor de temperatura'],
      hwOut: ['led A', 'led B', 'led C', 'led D', 'botao A', 'botao B', 'botao C', 'botao D', 'servo A', 'servo B', 'servo C', 'servo D'],
      leds: ['led A', 'led B', 'led C', 'led D'],
      outputs: ['ligado', 'desligado'],
      ops: ['>', '=', '<'],
      servos: ['servo A', 'servo B', 'servo C', 'servo D']
    }
};

tskauli, 2015-07-06:
Hello, I am the maintainer of the Norwegian (Bokmal) translation of Scratch. I am very happy and grateful for the Arduino extension and expect it to be widely used. Here is a Norwegian translation:

var blocks = { 
  nb: [
    ['h', 'når enheten tilkobles', 'whenConnected'],
    [' ', 'koble %m.hwOut til digital %n', 'connectHW', 'LED A', 3], 
    [' ', 'koble %m.hwIn til analog %n', 'connectHW', 'dreieknapp', 0], 
    ['-'],
    [' ', 'sett %m.leds %m.outputs', 'digitalLED', 'LED A', 'på'],
    [' ', 'sett %m.leds styrke til %n%', 'setLED', 'LED A', 100],
    [' ', 'endre %m.leds styrke med %n%', 'changeLED', 'LED A', 20],
    ['-'],
    [' ', 'rotér %m.servos til %n grader', 'rotateServo', 'servo A', 180],
    [' ', 'rotér %m.servos med %n grader', 'changeServo', 'servo A', 20],
    ['-'],
    ['h', 'når %m.buttons %m.btnStates', 'whenButton', 'knapp A', 'trykkes'],
    ['b', '%m.buttons trykket?', 'isButtonPressed', 'knapp A'],
    ['-'],
    ['h', 'når %m.hwIn %m.ops %n%', 'whenInput', 'dreieknapp', '>', 50],
    ['r', '%m.hwIn verdi', 'readInput', 'dreieknapp'],
    ['-'],
    [' ', 'sett digital %n %m.outputs', 'digitalWrite', 1, 'på'],
    [' ', 'set utgang %n til %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', 'når digital %n er %m.outputs', 'whenDigitalRead', 1, 'på'],
    ['b', 'digital %n på?', 'digitalRead', 1], 
    ['-'],
    ['h', 'når analog %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', 'analog %n verdi', 'analogRead', 0], 
    ['-'],
    ['r', 'skalér %n fra %n %n til %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    nb: {
      buttons: ['knapp A', 'knapp B', 'knapp C', 'knapp D'],
      btnStates: ['trykkes', 'slippes'],
      hwIn: ['dreieknapp', 'lyssensor', 'temperatursensor'],
      hwOut: ['LED A', 'LED B', 'LED C', 'LED D', 'knapp A', 'knapp B', 'knapp C', 'knapp D', 'servo A', 'servo B', 'servo C', 'servo D'],
      leds: ['LED A', 'LED B', 'LED C', 'LED D'],
      outputs: ['på', 'av'],
      ops: ['>', '=', '<'],
      servos: ['servo A', 'servo B', 'servo C', 'servo D']
    }
};

tskauli, 2015-07-06:
While I am at it, I can also offer a couple of comments to the English blocks:

- If we envisage a reader completely uninitiated to electronics then the meaning of "pin" may not be immediately obvious. A possible alternative is "digital 1" for digital I/O, "analog 1" for analog input and "output 1" for PWM output. It may or may not be better in English, but I went for this solution in the Norwegian translation here to sidestep the tricky translation of "pin".

- It may be an improvement to change the value block "read <rotation knob>" into "<rotation knob> value". Likewise, "read analog 0" could be changed into "analog 0 value".

Other than that, I think the Arduino blocks pallette is good, and I particularly like the very concrete "connect led A to pin 3" as an alternative to the more generic "set pin 3 on".

Thanks again for the work on an Arduino extension. I look forward to seeing the released version in Scratch!

Best regards,

  Torbjorn

khanning, 2015-07-22:
@wilsonjuniorr thank you for the Portuguese translation! It's been added to the extension.

@tskauli the Norwegian translation has been added. Thank you! 
I agree with your block suggestions. I think replacing "pin" with "digital" might simply things a little bit. The "rotation knob value" wording sounds better also.

if (rotation knob value) > 75
sounds better than
if (read rotation knob) > 75

Thanks for the suggestions!

@damellis what do you think about these changes to the English blocks?

damellis, 2015-07-22:
Personally, I think it would be confusing to say "digital 1" instead of "pin 1". I know that "pin" is a confusing word, and maybe we can find a better one, but I think it's good to use a word that refers to the thing we're talking about (i.e. a noun like "pin") rather than a word that describes the type of that thing (i.e. an adjective like "digital"). I'm open to alternatives like "connection" or "connector" or "input" or "output" but I'm not sure that any of them are really right. What do you think?

(I'm less concerned about the difference between "rotation knob value" and "read rotation knob". Probably there it's best to use whatever seems closest to other Scratch blocks.)

khanning, 2015-07-23:
@damellis makes a good point about noun vs. adjective. I think I still prefer "pin" to some of the other noun alternatives. Perhaps this will just have to be a learning opportunity about the term "pin". We can add a description to the documentation that explains what "pin" is referring to.

Currently, the analog blocks are using an adjective "analog", while the digital blocks are using a noun "pin". Should we change this wording to use an adjective and a noun (analog pin)?
[image]

ktprezes, 2015-07-24:
Hi everybody :)

this is polish translation:

var blocks = { 
  pl: [
    ['h', 'kiedy urządzenie jest podłączone', 'whenConnected'],
    [' ', 'podłącz %m.hwOut do pinu %n', 'connectHW', 'led A', 3], 
    [' ', 'podłącz %m.hwIn do we analogowego %n', 'connectHW', 'pokrętło', 0], 
    ['-'],
    [' ', 'ustaw %m.leds na %m.outputs', 'digitalLED', 'led A', 'włączony'],
    [' ', 'ustaw jasność %m.leds na %n%', 'setLED', 'led A', 100],
    [' ', 'zmień jasność %m.leds o %n%', 'changeLED', 'led A', 20],
    ['-'],
    [' ', 'obróć %m.servos w położenie %n degrees', 'rotateServo', 'serwo A', 180],
    [' ', 'obróć %m.servos o %n degrees', 'changeServo', 'serwo A', 20],
    ['-'],
    ['h', 'kiedy %m.buttons jest %m.btnStates', 'whenButton', 'przycisk A', 'wciśnięty'],
    ['b', 'czy %m.buttons jest wciśnięty?', 'isButtonPressed', 'przycisk A'],
    ['-'],
    ['h', 'kiedy %m.hwIn jest w położeniu %m.ops %n%', 'whenInput', 'pokrętło', '>', 50],
    ['r', 'odczytaj ustawienie %m.hwIn', 'readInput', 'pokrętła'],
    ['-'],
    [' ', 'ustaw pin %n jako %m.outputs', 'digitalWrite', 1, 'włączony'],
    [' ', 'ustaw pin %n na %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', 'kiedy pin %n jest %m.outputs', 'whenDigitalRead', 1, 'włączony'],
    ['b', 'czy pin %n jest włączony?', 'digitalRead', 1], 
    ['-'],
    ['h', 'kiedy we analogowe %n jest w położeniu %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', 'odczytaj we analogowe %n', 'analogRead', 0], 
    ['-'],
    ['r', 'przekształć wartość %n z zakresu %n %n na %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    pl: {
      buttons: ['przycisk A', 'przycisk B', 'przycisk C', 'przycisk D'],
      btnStates: ['wciśnięty', 'zwolniony'],
      hwIn: ['pokrętło', 'czujnik światła', 'czujnik temperatury'],
      hwOut: ['led A', 'led B', 'led C', 'led D', 'przycisk', 'przycisk B', 'przycisk C', 'przycisk D', 'serwo A', 'serwo B', 'serwo C', 'serwo D'],
      leds: ['led A', 'led B', 'led C', 'led D'],
      outputs: ['włączony', 'wyłączony'],
      ops: ['>', '=', '<'],
      servos: ['serwo A', 'serwo B', 'serwo C', 'serwo D']
    }
};

Anderson69s, 2015-07-30:
Here is the French Translation as promise on Twitter 🎉

<pre>var blocks = { 
  fr: [
    ['h', 'Quand l'appareil est connecté', 'whenConnected'],
    [' ', 'Connecté %m.hwOut au pin %n', 'connectHW', 'LED A', 3], 
    [' ', 'Connecté %m.hwIn au pin analogue %n', 'connectHW', 'Potentiomètre', 0], 
    ['-'],
    [' ', 'Régler %m.la LED %m.en Sortie', 'digitalLED', 'LED A', 'on'],
    [' ', 'Régler %m.la Luminosité de la LED à %n%', 'setLED', 'LED A', 100],
    [' ', 'Changer %m.la Luminosité de la LED de %n%', 'changeLED', 'LED A', 20],
    ['-'],
    [' ', 'Tourner %m.le Servo Moteur à %n degrés', 'rotateServo', 'Servo Moteur A', 180],
    [' ', 'Tourner %m.le Servo Moteur de %n degrés', 'changeServo', 'Servo Moteur A', 20],
    ['-'],
    ['h', 'Quand %m.le Bouton est %m.btnStates', 'whenButton', 'Bouton A', 'Appuyé'],
    ['b', 'Le %m.Bouton est-il pressé?', 'isButtonPressed', 'Bouton A'],
    ['-'],
    ['h', 'Quand %m.hwIn %m.ops %n%', 'whenInput', 'Potentiomètre', '>', 50],
    ['r', 'Lire %m.hwIn', 'readInput', 'Potentiomètre'],
    ['-'],
    [' ', 'Régler le Pin %n %m.en Sortie', 'digitalWrite', 1, 'On'],
    [' ', 'Régler le Pin %n à %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', 'Quand le Pin %n est %m.en Sortie', 'whenDigitalRead', 1, 'On'],
    ['b', 'Le Pin %n est-il démarré?', 'digitalRead', 1], 
    ['-'],
    ['h', 'Quand le Pin analogique est %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', 'Lire le Pin Analogique %n', 'analogRead', 0], 
    ['-'],
    ['r', 'Mapper %n de %n %n à %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    fr: {
      buttons: ['Bouton A', 'Bouton B', 'Bouton C', 'Bouton D'],
      btnStates: ['Appuyé', 'Relâché'],
      hwIn: ['Potentiomètre', 'Capteur de Lumière', 'Capteur de Temperature'],
      hwOut: ['LED A', 'LED B', 'LED C', 'LED D', 'Bouton A', 'Bouton B', 'Bouton C', 'Bouton D', 'Servo Moteur A', 'Servo Moteur B', 'Servo Moteur C', 'Servo Moteur D'],
      leds: ['LED A', 'LED B', 'LED C', 'LED D'],
      outputs: ['ON', 'OFF'],
      ops: ['>', '=', '<'],
      servos: ['Servo Moteur A', 'Servo Moteur B', 'Servo Moteur C', 'Servo Moteur D']
    }
}; </pre>

hackermesh, 2015-09-16:
Here is the spanish translation:

```javascript
var blocks = { 
  es: [
    ['h', 'al conectar el dispositivo', 'whenConnected'],
    [' ', 'conectar %m.hwOut al pin %n', 'connectHW', 'led A', 3], 
    [' ', 'conectar %m.hwIn al pin analógico %n', 'connectHW', 'rotation knob', 0], 
    ['-'],
    [' ', 'fijar estado de %m.leds a %m.outputs', 'digitalLED', 'led A', 'on'],
    [' ', 'fijar brillo de %m.leds a %n%', 'setLED', 'led A', 100],
    [' ', 'cambiar brillo de %m.leds por %n%', 'changeLED', 'led A', 20],
    ['-'],
    [' ', 'apuntar %m.servos en dirección %n grados', 'rotateServo', 'servo A', 180],
    [' ', 'girar %m.servos %n grados', 'changeServo', 'servo A', 20],
    ['-'],
    ['h', 'cuando el %m.buttons esté %m.btnStates', 'whenButton', 'button A', 'pressed'],
    ['b', '¿%m.buttons presionado?', 'isButtonPressed', 'button A'],
    ['-'],
    ['h', 'cuando %m.hwIn %m.ops %n%', 'whenInput', 'rotation knob', '>', 50],
    ['r', 'leer %m.hwIn', 'readInput', 'rotation knob'],
    ['-'],
    [' ', 'fijar estado de pin %n a %m.outputs', 'digitalWrite', 1, 'on'],
    [' ', 'fijar pin analógico %n al %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', 'cuando el pin %n esté %m.outputs', 'whenDigitalRead', 1, 'on'],
    ['b', '¿pin %n on?', 'digitalRead', 1], 
    ['-'],
    ['h', 'cuando pin analógico %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', 'leer analógico %n', 'analogRead', 0], 
    ['-'],
    ['r', 'convertir %n de %n %n a %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    es: {
      buttons: ['botón A', 'botón B', 'botón C', 'botón D'],
      btnStates: ['pulsado', 'liberado'],
      hwIn: ['potenciómetro', 'sensor de luz', 'sensor de temperatura'],
      hwOut: ['led A', 'led B', 'led C', 'led D', 'botón A', 'botón B', 'botón C', 'botón D', 'servo A', 'servo B', 'servo C', 'servo D'],
      leds: ['led A', 'led B', 'led C', 'led D'],
      outputs: ['on', 'off'],
      ops: ['>', '=', '<'],
      servos: ['servo A', 'servo B', 'servo C', 'servo D']
    }
};

khanning, 2015-09-21:
@hackermesh Thank you for the translation!

Anderson69s, 2015-09-22:
You're welcome!!!!

frankshieh, 2015-10-28:
Here is Traditional Chinese , Please note there are two types of Chinese, Traditional and Simplified which share the same code zh. But what's listed below is Traditional Chinese:
var blocks = { 
  zh: [
    ['h', '當裝置連接時', 'whenConnected'],
    [' ', '連接 %m.hwOut 到腳位 %n', 'connectHW', '發光二極體 A', 3], 
    [' ', '連接 %m.hwIn 到類比 %n', 'connectHW', '旋鈕', 0], 
    ['-'],
    [' ', '設定 %m.leds %m.outputs', 'digitalLED', '發光二極體 A', 'on'],
    [' ', '設定 %m.leds 亮度為 %n%', 'setLED', '發光二極體 A', 100],
    [' ', '改變 %m.leds 亮度 %n%', 'changeLED', '發光二極體 A', 20],
    ['-'],
    [' ', '旋轉 %m.servos 到 %n 度', 'rotateServo', '伺服馬達 A', 180],
    [' ', '旋轉 %m.servos %n 度', 'changeServo', '伺服馬達 A', 20],
    ['-'],
    ['h', '當 %m.buttons 為 %m.btnStates', 'whenButton', '按鈕 A', '按下'],
    ['b', '%m.buttons 按下?', 'isButtonPressed', '按鈕 A'],
    ['-'],
    ['h', '當 %m.hwIn %m.ops %n%', 'whenInput', '旋鈕', '>', 50],
    ['r', '讀取 %m.hwIn', 'readInput', '旋鈕'],
    ['-'],
    [' ', '設定腳位 %n %m.outputs', 'digitalWrite', 1, '開'],
    [' ', '設定腳位 %n 為 %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', '當腳位 %n 為 %m.outputs', 'whenDigitalRead', 1, '開'],
    ['b', '腳位 %n 開?', 'digitalRead', 1], 
    ['-'],
    ['h', '當類比 %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', '讀取類比 %n', 'analogRead', 0], 
    ['-'],
    ['r', '對應 %n 由 %n %n 為 %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    zh: {
      buttons: ['按鈕 A', '按鈕 B', '按鈕 C', '按鈕 D'],
      btnStates: ['按下', '放開'],
      hwIn: ['旋鈕', '光感應器', '溫度感應器'],
      hwOut: ['發光二極體 A', '發光二極體 B', '發光二極體 C', '發光二極體 D', '按鈕 A', '按鈕 B', '按鈕 C', '按鈕 D', '伺服馬達 A', '伺服馬達 B', '伺服馬達 C', '伺服馬達 D'],
      leds: ['發光二極體 A', '發光二極體 B', '發光二極體 C', '發光二極體 D'],
      outputs: ['開', '關'],
      ops: ['>', '=', '<'],
      servos: ['伺服馬達 A', '伺服馬達 B', '伺服馬達 C', '伺服馬達 D']
    }
};

khanning, 2016-02-02:
Korean (ko), Polish (pl), French (fr), and Traditional Chinese (zh) is now supported.

Thank you @neosarchizo, @ktprezes, @Anderson69s, and @frankshieh!

zharov, 2016-04-06:
Please add Russian translations. I have tried to test them in my fork, they have worked first time, then stopped to work for some unknown reason - but the translations themselves are correct.
var blocks={
  ru: [
      ['h', 'когда устройство подключено', 'whenConnected'],
      [' ', 'подключить %m.hwOut к выводу %n', 'connectHW', 'светодиод A', 3],
      [' ', 'подключить %m.hwIn к ан. входу %n', 'connectHW', 'потенциометр', 0],
      ['-'],
      [' ', 'установить %m.leds в %m.outputs', 'digitalLED', 'светодиод A', 'включен'],
      [' ', 'установить яркость %m.leds в %n%', 'setLED', 'светодиод A', 100],
      [' ', 'изменить яркость %m.leds на %n%', 'changeLED', 'светодиод A', 20],
      ['-'],
      [' ', 'установить %m.servos в позицию %n °', 'rotateServo', 'серво A', 180],
      [' ', 'повернуть %m.servos на %n °', 'changeServo', 'серво A', 20],
      ['-'],
      ['h', 'когда %m.buttons %m.btnStates', 'whenButton', 'кнопка A', 'нажата'],
      ['b', '%m.buttons нажата?', 'isButtonPressed', 'кнопка A'],
      ['-'],
      ['h', 'когда %m.hwIn %m.ops %n%', 'whenInput', 'потенциометр', '>', 50],
      ['r', 'значение %m.hwIn', 'readInput', 'потенциометр'],
      ['-'],
      [' ', 'установить выход %n в %m.outputs', 'digitalWrite', 1, 'включен'],
      [' ', 'установить ан. выход %n в %n%', 'analogWrite', 3, 100],
      ['-'],
      ['h', 'когда вход %n %m.outputs', 'whenDigitalRead', 1, 'включен'],
      ['b', 'вход %n вкл?', 'digitalRead', 1],
      ['-'],
      ['h', 'когда ан. вход %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
      ['r', 'значение ан. входа %n', 'analogRead', 0],
      ['-'],
      ['r', 'отобразить %n из %n %n в %n %n', 'mapValues', 50, 0, 100, -240, 240]
    ]
};

var menus = {
    ru: {
      buttons: ['кнопка A', 'кнопка B', 'кнопка C', 'кнопка D'],
      btnStates: ['нажата', 'отпущена'],
      hwIn: ['потенциометр', 'датчик света', 'датчик температуры'],
      hwOut: ['светодиод A', 'светодиод B', 'светодиод C', 'светодиод D', 'кнопка A', 'кнопка B', 'кнопка C', 'кнопка D', 'серво A', 'серво B', 'серво C', 'серво D'],
      leds: ['светодиод A', 'светодиод B', 'светодиод C', 'светодиод D'],
      outputs: ['включен', 'выключен'],
      ops: ['>', '=', '<'],
      servos: ['серво A', 'серво B', 'серво C', 'серво D']
    }
};

aytac, 2016-09-15:
Please add Turkish translate version of the blocks.

Thanks

   tr: [
      ['h', 'Cihaz bağlandığında', 'whenConnected'],
      [' ', '%m.hwOut yı pin %n e bağla', 'connectHW', 'led A', 3],
      [' ', '%m.hwIn yi analog %n e bağla', 'connectHW', 'rotation knob', 0],
      ['-'],
      [' ', '%m.leds yı %m.outputs olarak ayarla', 'digitalLED', 'led A', 'on'],
      [' ', '%m.leds nın parlaklığını % %n olarak ayarla', 'setLED', 'led A', 100],
      [' ', '%m.leds nın parlaklığını % %n değiştir', 'changeLED', 'led A', 20],
      ['-'],
      [' ', '%m.servos yu %n dereceye çevir', 'rotateServo', 'servo A', 180],
      [' ', '%m.servos yu %n derece değiştir', 'changeServo', 'servo A', 20],
      ['-'],
      ['h', '%m.buttons nın durumu %m.btnStates olduğunda', 'whenButton', 'button A', 'pressed'],
      ['b', '%m.buttons basılı ise', 'isButtonPressed', 'button A'],
      ['-'],
      ['h', '%m.hwIn %m.ops %n% olduğunda', 'whenInput', 'rotation knob', '>', 50],
      ['r', '%m.hwIn yı oku', 'readInput', 'rotation knob'],
      ['-'],
      [' ', 'Pin %n nın değerini %m.outputs yap', 'digitalWrite', 1, 'on'],
      [' ', 'Pin %n nın değerini %n% yap', 'analogWrite', 3, 100],
      ['-'],
      ['h', 'Pin %n in durumu %m.outputs olduğunda', 'whenDigitalRead', 1, 'on'],
      ['b', 'Pin %n in durumu açık ise', 'digitalRead', 1],
      ['-'],
      ['h', 'Analog %n in durumu %m.ops %n% olduğunda', 'whenAnalogRead', 1, '>', 50],
      ['r', 'analog %n i oku', 'analogRead', 0],
      ['-'],
      ['r', '%n değerini %n ile %n aralığından %n ile %n aralığına oranla', 'mapValues', 50, 0, 100, -240, 240]
    ],

----------------------------------------------------------

tr: {
      buttons: ['tuş A', 'tuş B', 'tuş C', 'tuş D'],
      btnStates: ['basıldı', 'bırakıldı'],
      hwIn: ['döndürme düğmesi', 'ışık sensörü', 'sıcaklık sensörü'],
      hwOut: ['led A', 'led B', 'led C', 'led D', 'tuş A', 'tuş B', 'tuş C', 'tuş D', 'servo A', 'servo B', 'servo C', 'servo D'],
      leds: ['led A', 'led B', 'led C', 'led D'],
      outputs: ['açık', 'kapalı'],
      ops: ['>', '=', '<'],
      servos: ['servo A', 'servo B', 'servo C', 'servo D']
    },

adlogi, 2017-03-28:
Here's the Arabic translation. Thanks!

var blocks = {
    ar: [
      ['h', 'عند وصل لوح الأردوينو', 'whenConnected'],
      [' ', 'صِل %m.hwOut إلى الرجل %n', 'connectHW', 'الثنائي الباعث أ', 3],
      [' ', 'صِل %m.hwIn إلى الرجل التماثلية %n', 'connectHW', 'مفتاح التدوير', 0],
      ['-'],
      [' ', 'اجعل %m.leds %m.outputs', 'digitalLED', 'الثنائي الباعث أ', 'في وضع التشغيل'],
      [' ', 'اجعل إضاءة %m.leds مساويةً %n%', 'setLED', 'الثنائي الباعث أ', 100],
      [' ', 'غير إضاءة %m.leds بمقدار %n%', 'changeLED', 'الثنائي الباعث أ', 20],
      ['-'],
      [' ', 'دوّر %m.servos إلى الموضع %n درجة', 'rotateServo', 'محرك السيرفو أ', 180],
      [' ', 'دوّر %m.servos بمقدار %n درجة', 'changeServo', 'محرك السيرفو أ', 20],
      ['-'],
      ['h', 'عندما يصبح %m.buttons %m.btnStates', 'whenButton', 'المفتاح أ', 'مضغوطًا'],
      ['b', '%m.buttons مضغوط؟', 'isButtonPressed', 'المفتاح أ'],
      ['-'],
      ['h', 'عندما يصبح %m.hwIn %m.ops %n%', 'whenInput', 'مفتاح التدوير', '>', 50],
      ['r', 'اقرأ %m.hwIn', 'readInput', 'مفتاح التدوير'],
      ['-'],
      [' ', 'اجعل الرجل %n %m.outputs', 'digitalWrite', 1, 'في وضع التشغيل'],
      [' ', 'اجعل الرجل %n مساويًة القيمة %n%', 'analogWrite', 3, 100],
      ['-'],
      ['h', 'عندما تصبح الرجل %n %m.outputs', 'whenDigitalRead', 1, 'في وضع التشغيل'],
      ['b', 'الرجل %n في وضع التشغيل؟', 'digitalRead', 1],
      ['-'],
      ['h', 'عندما تصبح الرجل التماثلية %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
      ['r', 'اقرأ الرجل التماثلية %n', 'analogRead', 0],
      ['-'],
      ['r', 'انقل %n من المجال %n %n إلى المجال %n %n', 'mapValues', 50, 0, 100, -240, 240]
    ]
  };

  var menus = {
    ar: {
      buttons: ['المفتاح أ', 'المفتاح ب', 'المفتاح ج', 'المفتاح د'],
      btnStates: ['مضغوطًا', 'محررًا'],
      hwIn: ['مفتاح التدوير', 'حساس الإضاءة', 'حساس الحرارة'],
      hwOut: ['الثنائي الباعث أ', 'الثنائي الباعث ب', 'الثنائي الباعث ج', 'الثنائي الباعث د', 'المفتاح أ', 'المفتاح ب', 'المفتاح ج', 'المفتاح د', 'محرك السيرفو أ', 'محرك السيرفو ب', 'محرك السيرفو ج', 'محرك السيرفو د'],
      leds: ['الثنائي الباعث أ', 'الثنائي الباعث ب', 'الثنائي الباعث ج', 'الثنائي الباعث د'],
      outputs: ['في وضع التشغيل', 'في وضع الإيقاف'],
      ops: ['>', '=', '<'],
      servos: ['محرك السيرفو أ', 'محرك السيرفو ب', 'محرك السيرفو ج', 'محرك السيرفو د']
    }
  };

scollovati, 2017-07-27:
Hi, 
I'm trying the Italian translation but it actually does not work. For istance a simple code like this one is not working.
[image]

I think that this is related to the translation of strings like _on/off_. The same happens for a simple button _pressed/released_.

Translation in other languages are working?

Babambabam, 2017-07-29:
Swedish translation :

var blocks = { 
  en: [
    ['h', 'när en apparat är inkopplad', 'whenConnected'],
    [' ', 'koppla %m.hwOut till pin %n', 'connectHW', 'led A', 3], 
    [' ', 'koppla %m.hwIn till analog %n', 'connectHW', 'ratt', 0], 
    ['-'],
    [' ', 'sätt %m.leds %m.outputs', 'digitalLED', 'led A', 'på'],
    [' ', 'sätt %m.leds styrka till %n%', 'setLED', 'led A', 100],
    [' ', 'ändra %m.leds styrka med %n%', 'changeLED', 'led A', 20],
    ['-'],
    [' ', 'rotera %m.servos till %n grader', 'rotateServo', 'servo A', 180],
    [' ', 'rotera %m.servos med %n grader', 'changeServo', 'servo A', 20],
    ['-'],
    ['h', 'när %m.buttons är %m.btnStates', 'whenButton', 'knapp A', 'nedtryckt'],
    ['b', '%m.buttons nedtryckt?', 'isButtonPressed', 'knapp A'],
    ['-'],
    ['h', 'när %m.hwIn %m.ops %n%', 'whenInput', 'ratt', '>', 50],
    ['r', 'read %m.hwIn', 'readInput', 'ratt'],
    ['-'],
    [' ', 'sätt pin %n %m.outputs', 'digitalWrite', 1, 'on'],
    [' ', 'sätt pin %n till %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', 'när pin %n är %m.outputs', 'whenDigitalRead', 1, 'på'],
    ['b', 'pin %n på?', 'digitalRead', 1], 
    ['-'],
    ['h', 'när analog %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', 'läs analog %n', 'analogRead', 0], 
    ['-'],
    ['r', 'sätt värden %n från %n %n to %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    en: {
      buttons: ['knapp A', 'knapp B', 'knapp C', 'knapp D'],
      btnStates: ['nedtryckt', 'släppt'],
      hwIn: ['ratt', 'ljus-sensor', 'temperatur-sensor'],
      hwOut: ['led A', 'led B', 'led C', 'led D', 'knapp A', 'knapp B', 'knapp C', 'knapp D', 'servo A', 'servo B', 'servo C', 'servo D'],
      leds: ['led A', 'led B', 'led C', 'led D'],
      outputs: ['på', 'av'],
      ops: ['>', '=', '<'],
      servos: ['servo A', 'servo B', 'servo C', 'servo D']
    }
};

NinjaTool, 2017-09-04:
Danish translation.

___.-{ BLOCKS }-.___

    da: [
      ['h', 'når Arduino er forbundet', 'whenConnected'],
      [' ', 'forbind %m.hwOut til pin %n', 'connectHW', 'LED A', 3],
      [' ', 'forbind %m.hwIn til analog %n', 'connectHW', 'drejeknap', 0],
      ['-'],
      [' ', 'sæt %m.leds %m.outputs', 'digitalLED', 'LED A', 'tændt'],
      [' ', 'sæt %m.leds lysstyrke til %n%', 'setLED', 'LED A', 100],
      [' ', 'ændr %m.leds lysstyrke med %n%', 'changeLED', 'LED A', 20],
      ['-'],
      [' ', 'roter %m.servos til %n grader', 'rotateServo', 'servo A', 180],
      [' ', 'roter %m.servos med %n grader', 'changeServo', 'servo A', 20],
      ['-'],
      ['h', 'når %m.buttons er %m.btnStates', 'whenButton', 'knap A', 'trykket'],
      ['b', '%m.buttons trykket?', 'isButtonPressed', 'knap A'],
      ['-'],
      ['h', 'når %m.hwIn %m.ops %n%', 'whenInput', 'drejeknap', '>', 50],
      ['r', 'læs %m.hwIn', 'readInput', 'drejeknap'],
      ['-'],
      [' ', 'sæt pin %n %m.outputs', 'digitalWrite', 1, 'tændt'],
      [' ', 'sæt pin %n to %n%', 'analogWrite', 3, 100],
      ['-'],
      ['h', 'når pin %n is %m.outputs', 'whenDigitalRead', 1, 'tændt'],
      ['b', 'pin %n tændt?', 'digitalRead', 1],
      ['-'],
      ['h', 'når analog %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
      ['r', 'læs analog %n', 'analogRead', 0],
      ['-'],
      ['r', 'placer %n fra %n %n mellem %n %n', 'mapValues', 50, 0, 100, -240, 240]
    ]

___.-{ MENUS }-.___

    da: {
      buttons: ['knap A', 'knap B', 'knap C', 'knap D'],
      btnStates: ['trykket', 'sluppet'],
      hwIn: ['dejeknap', 'lyssensor', 'temperatursensor'],
      hwOut: ['LED A', 'LED B', 'LED C', 'LED D', 'knap A', 'knap B', 'knap C', 'knap D', 'servo A', 'servo B', 'servo C', 'servo D'],
      leds: ['LED A', 'LED B', 'LED C', 'LED D'],
      outputs: ['tændt', 'slukket'],
      ops: ['>', '=', '<'],
      servos: ['servo A', 'servo B', 'servo C', 'servo D']
    }

mininis11, 2018-08-23:
czech translation (cs)

var blocks = { 
  cs: [
    ['h', 'when device is connected', 'whenConnected'],
    [' ', 'připojit %m.hwOut na pin %n', 'connectHW', 'led A', 3], 
    [' ', 'přípojit %m.hwIn na analogvoý pin %n', 'connectHW', 'otočný knoflík', 0], 
    ['-'],
    [' ', 'nastavit %m.leds na %m.outputs', 'digitalLED', 'led A', 'on'],
    [' ', 'nastavit jas %m.leds na %n%', 'setLED', 'led A', 100],
    [' ', 'změnit jas %m.leds o %n%', 'changeLED', 'led A', 20],
    ['-'],
    [' ', 'natoč %m.servos na %n stupňů', 'rotateServo', 'servo A', 180],
    [' ', 'otoč %m.servos o %n stupňů', 'changeServo', 'servo A', 20],
    ['-'],
    ['h', 'když %m.buttons je %m.btnStates', 'whenButton', 'tlačítko A', 'pressed'],
    ['b', '%m.buttons zmáčknuto?', 'isButtonPressed', 'tlačítko A'],
    ['-'],
    ['h', 'když %m.hwIn %m.ops %n%', 'whenInput', 'otočný knoflík', '>', 50],
    ['r', 'přečti %m.hwIn', 'readInput', 'otočný knoflík'],
    ['-'],
    [' ', 'nastavit pin %n na %m.outputs', 'digitalWrite', 1, 'on'],
    [' ', 'nastavit pin %n na %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', 'kdzž pin %n je %m.outputs', 'whenDigitalRead', 1, 'on'],
    ['b', 'je pin %n zapnut?', 'digitalRead', 1], 
    ['-'],
    ['h', 'když analog %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', 'přečti analog %n', 'analogRead', 0], 
    ['-'],
    ['r', 'mapuj %n od %n %n do %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    en: {
      buttons: ['tlačítko A', 'tlačítko B', 'tlačítko C', 'tlačítko D'],
      btnStates: ['zmáčknuto', 'nezmáčknuto'],
      hwIn: ['otočný knoflík', 'senzor světla', 'senzor teploty'],
      hwOut: ['led A', 'led B', 'led C', 'led D', 'tlačítko A', 'tlačítko B', 'tlačítko C', 'tlačítko D', 'servo A', 'servo B', 'servo C', 'servo D'],
      leds: ['led A', 'led B', 'led C', 'led D'],
      outputs: ['zapnuto', 'vypnuto'],
      ops: ['>', '=', '<'],
      servos: ['servo A', 'servo B', 'servo C', 'servo D']
    }
};

fengshuo2004, 2018-08-29:
Here is the **Simplified Chinese** Translation:
(If you want to credit me for the translation, please call me "David Feng")
```
var blocks = { 
  zh: [
    ['h', '当连接到设备时', 'whenConnected'],
    [' ', '连接 %m.hwOut 到端口 %n', 'connectHW', 'LED灯A', 3], 
    [' ', '连接 %m.hwIn 到模拟口 %n', 'connectHW', '旋钮', 0], 
    ['-'],
    [' ', '将 %m.leds 设置成 %m.outputs 电平', 'digitalLED', 'LED灯A', '高'],
    [' ', '将 %m.leds 的亮度设置到 %n%', 'setLED', 'LED灯A', 100],
    [' ', '将 %m.leds 的亮度增加 %n%', 'changeLED', 'LED灯A', 20],
    ['-'],
    [' ', '旋转伺服马达 %m.servos 到 %n 度', 'rotateServo', '伺服马达A', 180],
    [' ', '旋转伺服马达 %m.servos %n 度', 'changeServo', '伺服马达A', 20],
    ['-'],
    ['h', '当按钮 %m.buttons 被 %m.btnStates', 'whenButton', '按钮A', '按下'],
    ['b', '按钮 %m.buttons 被按下?', 'isButtonPressed', 'button A'],
    ['-'],
    ['h', '当 %m.hwIn 的模拟量 %m.ops %n%', 'whenInput', '旋钮', '>', 50],
    ['r', '读取 %m.hwIn 的模拟量', 'readInput', '旋钮'],
    ['-'],
    [' ', '将接口 %n 设成 %m.outputs 电平', 'digitalWrite', 1, '高'],
    [' ', '将模拟口 %n 设成 %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', '当接口 %n 的输入为 %m.outputs 电平', 'whenDigitalRead', 1, '高'],
    ['b', '接口 %n 为高电平?', 'digitalRead', 1], 
    ['-'],
    ['h', '当模拟口 %n 的输入 %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', '读取模拟口 %n 的输入', 'analogRead', 0], 
    ['-'],
    ['r', '映射数值 %n 从 %n %n 到 %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    zh: {
      buttons: ['按钮A', '按钮B', '按钮C', '按钮D'],
      btnStates: ['按下', '放开'],
      hwIn: ['旋钮', '光线传感器', '温度传感器'],
      hwOut: ['LED灯A', 'LED灯B', 'LED灯C', 'LED灯D', '按钮A', '按钮B', '按钮C', '按钮D', '伺服马达A', '伺服马达B', '伺服马达C', '伺服马达D'],
      leds: ['LED灯A', 'LED灯B', 'LED灯C', 'LED灯D'],
      outputs: ['高', '低'],
      ops: ['>', '=', '<'],
      servos: ['伺服马达A', '伺服马达B', '伺服马达C', '伺服马达D']
    }
}; 
```

Jennice16, 2019-01-05:
Hi all, Has this project been abandoned? I think it's a great project, but it seems that danish translation (lang=da) has not been implemented? Or am I doing something wrong?

NinjaTool, 2019-01-06:
Did I send the translations in a wrong format?

Jennice16, 2019-01-06:
It seems the originator has not edited the thread for the last 2 years.

NinjaTool: If you're familiar with GitHub and how the branching works (I'm a noob at this), please try to follow the originally suggested 3 steps (adding to the .js file) by adding the translation sections to the menus and blocks sections, respectively, and make the "pull request" (step 3), whatever that means...

It could be SO cool if this could be translated to danish, but I don't have the IT knowledge to do it.

uemil, 2019-02-13:
Here's the Romanian translation.

var blocks = { 
  en: [
    ['h', 'când Arduino este conectat', 'whenConnected'],
    [' ', 'conectează %m.hwOut la pinul %n', 'connectHW', 'led A', 3], 
    [' ', 'conectează %m.hwIn la pinul analog %n', 'connectHW', 'butonul rotativ', 0], 
    ['-'],
    [' ', 'setează %m.leds %m.outputs', 'digitalLED', 'led A', 'on'],
    [' ', 'setează %m.leds luminozitatea la %n%', 'setLED', 'led A', 100],
    [' ', 'schimbă %m.leds luminozitatea cu %n%', 'changeLED', 'led A', 20],
    ['-'],
    [' ', 'învârte %m.servos la %n grade', 'rotateServo', 'servo A', 180],
    [' ', 'învârte %m.servos cu %n grade', 'changeServo', 'servo A', 20],
    ['-'],
    ['h', 'când %m.buttons este %m.btnStates', 'whenButton', 'button A', 'pressed'],
    ['b', '%m.buttons apasat?', 'isButtonPressed', 'button A'],
    ['-'],
    ['h', 'când %m.hwIn %m.ops %n%', 'whenInput', 'rotation knob', '>', 50],
    ['r', 'citește %m.hwIn', 'readInput', 'rotation knob'],
    ['-'],
    [' ', 'setează pin %n %m.outputs', 'digitalWrite', 1, 'on'],
    [' ', 'setează pin %n to %n%', 'analogWrite', 3, 100],
    ['-'],
    ['h', 'când pin %n is %m.outputs', 'whenDigitalRead', 1, 'on'],
    ['b', 'pinul %n este pornit?', 'digitalRead', 1], 
    ['-'],
    ['h', 'când pinul analog %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
    ['r', 'citește pinul analog %n', 'analogRead', 0], 
    ['-'],
    ['r', 'mapează %n de la %n %n la %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    en: {
      buttons: ['buton A', 'buton B', 'buton C', 'buton D'],
      btnStates: ['apăsat', 'neapăsat'],
      hwIn: ['buton rotativ', 'senzor de lumină', 'senzor de temperatură'],
      hwOut: ['led A', 'led B', 'led C', 'led D', 'buton A', 'buton B', 'buton C', 'buton D', 'servo A', 'servo B', 'servo C', 'servo D'],
      leds: ['led A', 'led B', 'led C', 'led D'],
      outputs: ['pornit', 'oprit'],
      ops: ['>', '=', '<'],
      servos: ['servo A', 'servo B', 'servo C', 'servo D']
    }
};

KY-Leng, 2019-12-04:
var blocks = { 
  khm: [
	['h', 'ពេលភ្ជាប់ឧបករណ៏រួច', 'whenConnected'],
	[' ', 'ភ្ជាប់ %m.hwOut ទៅជើងលេខ %n', 'connectHW', 'អំពូល LED A', 3], 
	[' ', 'ភ្ជាប់ %m.hwIn ទៅច្រកអាណាឡូកលេខ %n', 'connectHW', 'ប៉ូតង់ស្យូមម៉ែត្រ', 0], 
    ['-'],
	[' ', 'កំណត់ %m.leds %m.outputs', 'digitalLED', 'អំពូល LED A', 'បើក'],
	[' ', 'កំណត់កំរិតពន្លឺរបស់ %m.leds ត្រឹម %n%', 'setLED', 'អំពូល LED A', 100],
	[' ', 'ប្តូរកំរិតពន្លឺរបស់ %m.leds ចំនួន %n%', 'changeLED', 'អំពូល LED A', 20],
	['-'],
	[' ', 'បង្វិល %m.servos ទៅត្រឹម %n ដឺក្រេ', 'rotateServo', 'ម៉ូទ័រស៊ែវ៉ូ A', 180],
	[' ', 'បង្វិល %m.servos ចំនួន %n ដឺក្រេ', 'changeServo', 'ម៉ូទ័រស៊ែវ៉ូ A', 20],
    ['-'],
	['h', 'នៅពេល %m.buttons ត្រូវបាន %m.btnStates', 'whenButton', 'ប៉ូតុង A', 'ចុច'],
	['b', '%m.buttons ចុច?', 'isButtonPressed', 'ប៉ូតុង A'],
	['-'],
	['h', 'នៅពេលតំលៃ %m.hwIn %m.ops %n%', 'whenInput', 'ប៉ូតង់ស្យូមម៉ែត្រ', '>', 50],
	['r', 'ទាញយកតំលៃ %m.hwIn', 'readInput', 'ប៉ូតង់ស្យូមម៉ែត្រ'],
	['-'],
	[' ', 'កំនត់ច្រកលេខ %n %m.outputs', 'digitalWrite', 1, 'បើក'],
	[' ', 'កំនត់តំលៃច្រកលេខ %n ត្រឹម %n%', 'analogWrite', 3, 100],
	['-'],
	['h', 'នៅពេលជើងលេខ %n ត្រូវបាន %m.outputs', 'whenDigitalRead', 1, 'បើក'],
	['b', 'ជើងលេខ %n បើក?', 'digitalRead', 1], 
	['-'],
	['h', 'នៅពេលតំលៃច្រកលេខ %n %m.ops %n%', 'whenAnalogRead', 1, '>', 50],
	['r', 'ទាញយកតំលៃច្រកលេខ %n', 'analogRead', 0], 
	['-'],
	['r', 'ភ្ជាប់តំលៃ %n ពីចន្លោះ %n %n ទៅចន្លោះ %n %n', 'mapValues', 50, 0, 100, -240, 240]
  ],
};

var menus = {
    en: {
      buttons: ['ប៉ូតុង A', 'ប៉ូតុង B', 'ប៉ូតុង C', 'ប៉ូតុង D'],
      btnStates: ['ចុច', 'មិនទាន់ចុច'],
      hwIn: ['ប៉ូតង់ស្យូមម៉ែត្រ', 'សិនស័រពន្លឺ', 'សិនស័រសីតុណ្ហភាព'],
      hwOut: ['អំពូល LED A', 'អំពូល LED B', 'អំពូល LED C', 'អំពូល LED D', 'ប៉ូតុង A', 'ប៉ូតុង B', 'ប៉ូតុង C', 'ប៉ូតុង D', 'ម៉ូទ័រស៊ែវ៉ូ A', 'ម៉ូទ័រស៊ែវ៉ូ B', 'ម៉ូទ័រស៊ែវ៉ូ C', 'ម៉ូទ័រស៊ែវ៉ូ D'],
      leds: ['អំពូល LED A', 'អំពូល LED B', 'អំពូល LED C', 'អំពូល LED D'],
      outputs: ['បើក', 'បិទ'],
      ops: ['>', '=', '<'],
      servos: ['ម៉ូទ័រស៊ែវ៉ូ A', 'ម៉ូទ័រស៊ែវ៉ូ B', 'ម៉ូទ័រស៊ែវ៉ូ C', 'ម៉ូទ័រស៊ែវ៉ូ D']
    }
};

PERCE-NEIGE, 2020-06-24:
I came here to offer a French translation, but I see there's already one. So, the French language will be added soon? (because the message has been posted in 2015!)

PERCE-NEIGE, 2020-06-24:
@Jennice16 So, this page has been abandoned, but not the translation via pull requests?

PERCE-NEIGE, 2020-06-24:
"Korean (ko), Polish (pl), French (fr), and Traditional Chinese (zh) is now supported."

I don't see French in the project page.

SparkScrach-P, 2020-11-09:
I will surely give an extension for **bengali** within a week...just tell if it already exists or not

SparkScratch-P, 2021-09-03:
> I will surely give an extension for **Bengali** within a week...just tell if it already exists or not

No reply yet!