khanning/scratch-arduino-extension #16: Integrated resistors

christianimmler, 2015-05-24:
Hi,
The Scratch extension seems to switch on the integrated Pulldown resistors on Arduino input pins. I am working on a project with sensor contacts like MakeyMakey http://www.makeymakey.com/. There I cannot use any integrated resistors. Is there a way to switch them of in the Scratch extension?

damellis, 2015-05-24:
Hmm... can you explain a bit more about what you're seeing (e.g. the circuit, code, behavior)? As far as I know, there aren't pull-down resistors on the Arduino (only built-in pull-up resistors which I don't think we enable).

christianimmler, 2015-05-24:
[image]
This is the circuit, I am using. The touch sensor ('Kontaktsensor' in the drawing) is connected via 20 MOhm resistor to +3.3V and directly to the Arduino pin. So the input pin on Arduino ('GPIO' in the drawing) gets a week but defined high signal. When a person who is always grounded, touches the sensor the signal will go to low.
This is working well with normal Arduino code when NOT using any resistor on the Arduino. The Scratch extension does not recognize any change when touching the sensor or not.

damellis, 2015-05-24:
Hmm... what value do you get for the sensor in Scratch, high or low? Have you tried with a smaller resistor (e.g. 1M instead of 22M)?

dmellis, 2015-05-24:
Please remove me from all of your lists.

christianimmler, 2015-05-25:
[image]
Hi, here a simple sample. When touching the sensor the leds should be switched.
But... led A is always on, so pin 2 connected to the sensor seems to be always high and does not change its state, when the sensor is touched. The same effect with 10M resistor.
When connecting the sensor with a wire to GND, led A goes off and led B goes on. I conclude, the logic in my script is ok.

damellis, 2015-05-25:
Hmm, a quick look at the StandardFirmata code makes it look like it doesn't set the pull-up resistors. Kreg, we don't set them in the extension, do we?

What if you try touch pin 2 to ground through different resistors (e.g. 10K, 100K, 1M, 10M)? Which ones successfully trigger the toggling of your LEDs?

khanning, 2015-05-26:
The ```<pin [2] on?>``` block sets the pin mode to INPUT so the integrated pull-up resistors should not be enabled. 

It looks like the only way to enable the pull-up resistors with StandardFirmata is to set the pin mode to INPUT and then set the pin HIGH. This is not currently possible in the extension.
http://www.firmata.org/wiki/Proposals#Query_Pull-up_Resistor_Status_Proposal