khanning/scratch-arduino-extension #19: Arduino not connected message in ScratchX

robbieberg78, 2015-06-17:
I get as far as loading the Scratch Arduino extension into ScratchX (running Safari 8.0.6, OS X 10.10.3, Arduino Uno running standard Firmata), where I can see the new blocks, but not communicate with the Arduino. (Status light remains yellow,  Arduino disconnected message). This problem is not unique to the Scratch Arduino extension  but rather stems from a general inability to find the serial port from ScratchX.  For example, a similar thing happens when I try to load the PicoBoard extension in ScratchX, even though PicoBoards work fine when I load the extension from Scratch 2.0. 

When I try using Firefox or Chrome I don't even get this far - I can't even successfully  get the Scratch Browser plug-in to load in with of those browsers. I haven't tried a Windows machine yet.

I'd be grateful for any suggestions you might have. This work is  part of a project that I'm working on with LLK (with Mitchel, Natalie, and Tim). Thanks!

Robbie Berg

damellis, 2015-06-17:
I'm not sure. That sounds similar to my setup and I haven't had any trouble. Are you around LLK? We can try to debug in person. Also, you might try asking the Scratch team (e.g. on the ScratchX github: https://github.com/LLK/scratchx/issues) as I don't have much insight into how the browser plugin works.

robbieberg78, 2015-06-17:
Hi David,

Thanks for getting back to me so quickly. I agree that it's likely a
browser plug-in related issue.  I suspect Sayamindu is the right person to
ask about this.

By way of background:  we've been working on creating a programmable
version of Mike Petrich and Karen Wilkinson's beautiful Light Play activity
(vimeo.com/115030170) for use with young children. Our short term goal is
to get a small number of prototypes ready for us to play with during an
internal design charrette that we're planning for early August at the
Exploratorium in SF with Mike, Karen, Tim, Natalie, and myself. Our primary
focus has been on using Scratch Jr. as the programming environment,
connected via Bluetooth or Wifi to the hardware (an Arduino+ custom shield)
that controls the lights and motors. The bandwidth requirements are very
low, we just need to send a few bytes back and forth each second. For the
design charette it would be nice if we could talk to the hardware via Scratch
as well, since we'd like to be able to iterate quickly on ideas for the
software design. All we really need for this is a ScratchX extension with
two custom blocks: one to send a serial byte to the Arduino and one that
can receive a serial byte.

I should be around LLK in a week or so to get together with Tim and
Natalie. If you're around I'll stop by; Mitchel mentioned that you would be
a good person to talk to about this project.

Best,
Robbie

khanning, 2015-06-17:
Hi Robbie,

When you open the extension on ScratchX and connect the Arduino are you seeing any messages in the JavaScript console? There should be a message ```Attempting connection with /dev/ttyUSB***``` for each serial port the browser plug-in is able to identify.

In Chrome, if you go to chrome://plugins/ is the Scratch Device Plugin listed?
Same with Firefox on the about:plugins page?

On Mac, the browser plug-in is looking for ports in /dev that begin with tty.usbmodem, tty.serial, or tty.usbserial. Can you check which serial port your Arduino is using when it's connected?

robbieberg78, 2015-06-18:
Hi Kreg,

Thanks for your quick response.

As you can see from the Chrome screen shot below I do indeed see  a
message Attempting
connection with /dev/tty.usbmodem1421 when I plug in an Arduino Uno. The
message keeps refreshing and the status light remains yellow. I never see
any activity on the Arduino's Rx and Tx LEDs.

In the Arduino Tools menu it lists the port as cu.usbmodem1421. I have only
a vague understanding of the distinction between a tty and a cu serial
connection, but I'm guessing that this is not the source of my difficulty.

By the way, I was able to get the Scratch Extensions plug-in in Chrome to
work after I realized that I had previously neglected to enable  NPAPI
plug-ins.

If you have any further suggestions I'd love to hear them.

Thanks!, Robbie

[image: Inline image 1]

khanning, 2015-06-18:
The screenshot didn't seem to come through in your last comment.

I set up another copy of the Arduino extension that has more logging enabled. Would you mind trying this version and sending me a log of the the JavaScript console? 
http://scratchx.org/?url=http://khanning.github.io/scratch-arduino-extension/arduino_extension_debug.js#scratch

Could you also try loading this sample project in the official version of Scratch? It has the Arduino extension already attached to it. If the extension still doesn't connect please send in the console log from this attempt.
[Arduino Debug Test.sb2](https://drive.google.com/open?id=0BzFuPSpksnPrdGtzbjRYX0hIVFE&authuser=0)

Thank you!

robbieberg78, 2015-06-18:
Hi Kreg,

The ScratchX extension s failing to open the serial connection so nothing
shows up in the Arduino Debug window. Brian Silverman and I spoke with
Sayamindu and he's pretty sure the problem is with the ScratchX plugin. I'm
getting together with him tomorrow at the Media Lab to try to figure it
out. I'll keep you posted...

Thanks for your help with this.

Robbie Berg
[image: Inline image 1]

kaspesla, 2015-06-30:
Any update on this? I'm trying to make a plugin that uses serial and having no luck. Using your logging extension above, I get no console output at all.  In my other tests, _deviceConnected is not getting called at all. I've tried Safari and Chrome on 10.8 and 10.10. Thanks!

technodream, 2015-09-27:
Hi 
Just adding details - I have the same issue, 
connecting arduino (port COM5 in device manager) ,Arduino IDE can communicate well.
connecting via ScratchX I see in the Javascript console "attempting connectiuon with COM5" over and over. by the looks of the RX/TX/13 leds the Arduino seems to reset itself. 
seems almost like mismatch in communication speed

thanks for your good work! looking forward to using this extension in my courses

[image]

christianimmler, 2016-01-10:
Hi,
I have the same problem with Chrome on Windows. Arduino IDE and the Firmata Test program are working well on COM3. Chrome shows:

Attempting connection with COM3  arduino_extension.js:506

Chrome and the plugin helper have been installed today with the latest versions.

christianimmler, 2016-01-10:
I tried it on another PC with Firefox. Arduino IDE and the Firmata Test program are working well on COM4. Firefox shows:
[image]

The connection symbol is green, but errors in the console window. he LED does not blink.

sinanthottan, 2016-04-27:
Hello,

I am facing the same issue with chrome on windows. The console shows attempting connection with COM7, but the light is still yellow.

With Mozilla on windows, was not able to open scratchx but now it works when installed latest Flash Player.
Scratchx and the LED projects works well now with Mozilla on WIndows.

PassivAgresHunt, 2016-07-12:
HI, I am trying to access the littleBits extention with Scratch and the Browser Plug-in link is broken. Does anyone have another solution or another link for the broswer plug in?

yckelvin, 2017-10-05:
Hi all, 

I am a newbie to Arduino and ScratchX, my configuration is macOS Sierra with ver. 10.12.5 (16F73), Arduino UNO. I can connect the Arduino with Arduino IDE with the problem. 

I closed the Arduino IDE and ensure the connection is closed. I check the Chrome plugins with chrome://extensions/ and Scratch Device Plugin Helper 2.2.1.0 is enabled after I download and install the plugins from https://cdn.scratch.mit.edu/scratchr2/static/__228c049f0e0c9fa4e6c6c9cbe016c00e__/ext/ScratchDevicePluginChrome.dmg

My Chrome is Version 61.0.3163.100 (Official Build) (64-bit), I launch the online ScratchX by clicking http://scratchx.org/?url=http://khanning.github.io/scratch-arduino-extension/arduino_extension.js, the Arduino board cannot be connected, from Console, it keeps Attempting connection with /dev/tty.usbmodem1421. 

How can I do further troubleshooting? 

Best regards,

Kelvin.
[image]
[image]