khanning/scratch-arduino-extension #20: Unable to connect to Arduino - windows

DeeJay, 2015-06-21:
This is a near duplicate of Robbie Berg's report in Issue #19, but with the important difference that my OS is Windows 7 Home Premium SP1. I have chosen to open another issue so as not to 'pollute' or confuse the existing discusssion, but obviously if you want to close or merge this report because of their similarity you should do so.

I have tried using both Firefox and Chrome, installing and enabling the appropriate plugins and helpers.

In all cases, the extension appears to load and creates the Arduino label under More Blocks, but the status indicator remains Yellow. The logging console repeatedly logs "Attempting connection with COM4". [This comes from the function tryNextDevice() ]  I think it might be significant that at this point my system flat-lines with 100% cpu usage - my hunch is that the code is spinning hard on a lock or some other resource conflict.

Loading the _debug version of the extension (as advised in #19) gives the same result.

Running the standard online Scratch2 editor and loading a locally saved copy of Arduino Debug Test.sb2 (again as referenced in #19) results in the extension loading AND giving a Green status indicator. But it does not give a working solution: dragging SET PIN x ON blocks onto the stage and clicking them does not actually change the LED status.

The tail of the log is as follows -
Uint8Array [ 121, 2, 3, 47, 0, 109, 0, 110, 0, 116, 4086 more… ] d347dd92d983e6c6eeb44cd1b206aa6a:180:4
"Input Data:" d347dd92d983e6c6eeb44cd1b206aa6a:511:6
Uint8Array [ 240, 121, 2, 3, 47, 0, 109, 0, 110, 0, 95 more… ] d347dd92d983e6c6eeb44cd1b206aa6a:512:6
"Sysex Message:" d347dd92d983e6c6eeb44cd1b206aa6a:179:4
Uint8Array [ 121, 2, 3, 47, 0, 109, 0, 110, 0, 116, 4086 more… ] d347dd92d983e6c6eeb44cd1b206aa6a:180:4
"Input Data:" d347dd92d983e6c6eeb44cd1b206aa6a:511:6
Uint8Array [ 240, 121, 2, 3, 47, 0, 109, 0, 110, 0, 95 more… ] d347dd92d983e6c6eeb44cd1b206aa6a:512:6
"Sysex Message:" d347dd92d983e6c6eeb44cd1b206aa6a:179:4
Uint8Array [ 121, 2, 3, 47, 0, 109, 0, 110, 0, 116, 4086 more… ] d347dd92d983e6c6eeb44cd1b206aa6a:180:4
"Error opening serial device COM4: TypeError: r is null" project_base.js:8:5201
"Attempting connection with COM4" d347dd92d983e6c6eeb44cd1b206aa6a:508:4

To try to eliminate as many other factors as possible:
I have checked my serial port (CP210x) and arduino (a 'shrimp clone) by using codebender to upload a basic 'blink' sketch, and then replacing it with the StandardFirmata sketch. I have checked the installation of the Firmata sketch by running a trivial python script from the pymata package which blinks the on-board LED.

Unlike your other correspondent I am not working geographically close to you: I am 'over the pond' in Leicester, UK, so I am not familiar with any of the other named collaborators in this project.

khanning, 2015-06-22:
Hi @DeeJay, thanks for reporting the issue!

#19 issue was caused by the latest version of the Scratch Browser Plug-in. Serial devices resumed working properly after reverting back to the previous version of the plug-in.

I think I still have the [previous version of the Windows plug-in](https://drive.google.com/open?id=0BzFuPSpksnPrd21tUkZXUmE4dHM&authuser=0) cached if you would like to try it. I would remove your current plug-in through "Add/Remove Programs" before reinstalling this just to be sure.

Please let us know if this works for you.

DeeJay, 2015-06-22:
Thank you for your prompt reply.

I am pleased to confirm that V1.0.0 of the Scratch Browser Plugin resolves this issue for both Firefox and Chrome. (I had previously installed V2.0.107.0 of the plugin by following the links in the scratch-arduino-extension documentation.)

closed 2015-06-22 by DeeJay