khanning/scratch-arduino-extension #23: Adding new translations?

damellis, 2015-07-21:
Looks like we have a few new translations (Japanese and Italian in pull requests) and Norwegian and Portuguese in issue #15. Do you have time to incorporate these? Anything I can do to help?

khanning, 2015-07-22:
Added new translations in commit 8e2cbea

closed 2015-07-22 by khanning

damellis, 2015-07-22:
Awesome! Do you want to commit them to the master branch, too?

khanning, 2015-07-23:
My mistake, meant to do that at the same time.

What do you think about making the gh-pages the default branch and deleting the master branch all together?

damellis, 2015-07-23:
Up to you. It's a bit unconventional, but given that all the code is in the Javascript file that's in the gh-pages branch anyway, it seems reasonable enough.