khanning/scratch-arduino-extension #24: Adding checkboxes to reporter blocks (to show values in stage)?

damellis, 2015-07-23:
Is this possible (e.g. on the "read rotation knob" block and the "read analog 0" block)? Can we do it on the boolean blocks ("button A pressed" and "pin 1 on") too?

khanning, 2015-07-28:
This is not currently possible with Scratch extensions. Checkboxes are only visible for reporter blocks that contain no arguments. We could break out each of the components in the "read rotation knob" block into their own discrete block, but this would result in a much longer block list.

Boolean blocks don't get the checkbox whether they have arguments or not. It looks like none of the official Scratch boolean blocks get the checkbox either. The feature must not be implemented for booleans.

I opened an issue on the ScratchX repo about this.
https://github.com/LLK/scratchx/issues/44

damellis, 2015-07-29:
Ah, thanks for the info!