khanning/scratch-arduino-extension #29: ScratchX Arduino not connecting (Mac+Chrome browser)

RRWagner, 2015-09-11:
I have installed the plug-ins and the animated LED and Pico sprite work, but there is no successful connection to the Arduino that has Standard Firmata installed on it.  In the Chrome browser console, it displays:

attempting connection with /dev/tty.usbmodemfa131 arduino_extension.js:506 

where the link for the extension is: 

http://scratchx.org/?url=http://khanning.github.io/scratch-arduino-extension/examples/Arduino%20Example%20-%20Blink%20an%20LED.sbx

The Extensions list shows:

 Scratch Device Plugin Helper 2.1.2.0

which says that it also requires the Scratch Device Plugin, which I did install.  However, Google also says that after Sept. 1, 2015 NPAPI plug-ins are not supported:

https://support.google.com/chrome/answer/6213033?hl=en

Has anyone gotten ScratchX Arduino browser working in Chrome on a Mac since Sept. 1, 2015?

khanning, 2015-09-28:
@RRWagner thank you for reporting this issue. It looks like there is currently an problem with the way serial device data is read using the Scratch Device Plugin for Chrome. This issue occurs on both Mac and Windows with the Chrome browser.

As a temporary workaround, you can use Firefox and the [Scratch plugin for "other web browsers"](https://scratch.mit.edu/info/ext_download/)

cactrot, 2015-10-22:
Work in progress, but there is hope :)

https://github.com/cactrot/scratch-device-host-chrome-app

davidheld, 2015-12-05:
Any update on this issue? I was hoping to run this on a Chromebook? BTW, thanks for creating this powerful utility!

davidheld, 2015-12-24:
So I can use Firefox now but ...."Mozilla intends to remove support for most NPAPI plugins in Firefox by the end of 2016." So then what?

fluffware, 2016-01-24:
I'm working on a Chrome plugin for Linux and found that one of the problems is that the Arduino extension tries to call set_receive_handler directly after the call to open. It should supply a callback to open and register the handler there. The [Pico extension](https://github.com/LLK/scratchx/blob/gh-pages/scratch_extensions/picoExtension.js) do it correctly.

GrandWizzardTheodore, 2016-04-22:
Look forward to getting this working! It would be a great thing to have. My son wants his Lego Wedo back so I need to use the Arduino to make servos work and expand his work with lights. 

I tried to get Firefox to work but that refused as well for me. Safari was a blank and new microsoft explorer? didn't seem interested.

HyperDuino, 2016-04-23:
Hello.  It's been about six months since I asked about this.  I do presentations at educational conferences, and I'd like to be able to show Arduino for ScratchX, but I've yet to get it working.  I spent a couple of hours tonight trying things, and documenting the process for teachers, but in the end it's a dead-end.  I'm using a Chrome browser on Mac OS X 10.7.5. 

Is there anything that can be done to make this work on a Mac with Chrome browser?  Does this currently work in anything other than Windows?

Thank you,

Roger

khanning, 2016-04-26:
There is currently a bug in the Scratch Extensions Browser Plugin that prevents the Arduino extension from working properly in the Chrome browser for both Mac and Windows. In the meantime, you can use the Firefox browser along with the [Scratch Extensions Browser Plugin](https://scratch.mit.edu/info/ext_download/) for "other web browsers". The Arduino extension should work properly with this setup. Please let me know if this does not work for you. 

I will update this issue once the Chrome bug has been fixed.

Thank you!

HyperDuino, 2016-04-26:
Thank you for the response.  I'm limited to Chromebooks, so this will be great when the Chrome bug gets fixed.

cactrot, 2016-04-26:
I think there might be more than a bug in the plugin. The new flash plugin for Chrome is built on the PPAPI. I can not find anything in the PPAPI docs that mention serial port and filesystem access is sandboxed. The only solution I have found is the Chrome.serial api.

HyperDuino, 2016-04-28:
When I install the Firefox extension and plug-in on Mac OS X, I can see the ScratchX plug-in, but I don't see the Arduino extension in the list. However, I do get the green dot next to "Arduino" and things seem to work.  Is the extension somehow hidden?

PassivAgresHunt, 2016-07-12:
HI, I am trying to access the littleBits extention with Scratch and the Browser Plug-in link is broken. Does anyone have another solution or another link for the broswer plug in?

yckelvin, 2017-11-03:
I launch the Arduino Extension with Scratch on macOS (v10.12.5) with Chrome (v61.0.3163.100) but cannot turn on the Arduino, there is no problem to Arduino Web editor. Is the extension already stop support?

x35gaming, 2018-11-12:
chrome wont even open the plugins