khanning/scratch-arduino-extension #39: Indicator under "More Blocks" stays yellow, disconnected, Chrome, Windows 8.1

Brasshopper, 2016-04-02:
I am at a loss about what to do next.  I have installed both halves of the browser extension.  I ran a simple blink program to verify that pin 9 of the Arduino board was correctly connected to the LED.  I then installed the StandardFirmata from the examples library.  This seems to have worked.
But the indicator stays yellow.  Opening the javascript console reveals these messages. 

59 Attempting connection with COM3
arduino_extension.js:521 Uncaught TypeError: Cannot read property 'set_receive_handler' of null
2 arduino_extension.js:508 Attempting connection with COM3
48 arduino_extension.js:508 Attempting connection with COM3
arduino_extension.js:508 Attempting connection with COM3
230 arduino_extension.js:508 Attempting connection with COM3

COM3 is, in fact, the correct serial device.  It is the one I use in the Arduino software.  Furthermore, when the scratchx extension is loaded, the transmit and receive lights are flashing on the board, more or less constantly, several times per 10 seconds.  If I kill that tab, the lights stop flashing.  

If I then click on http://scratchx.org/?url=http://khanning.github.io/scratch-arduino-extension/arduino_extension.js#scratch then the transmit and receive conversation starts again.  The pin 13 light flashes. The RX and TX lights flash.  The indicator on the page stays yellow.  

So SOMETHING that is associated with that tab has com3 open and is talking to it.  But whatever that thing is, it is not talking to the indicator on the page.  It stays yellow and reports disconnected.

In a couple trys, I can get the port monitor included with the arduino software open.  Resetting the rate to 57600 gives

S t a n d a r d F i r m d a t a . i n o ÷

I think the spaces are nulls, I can't cut and paste them.

Reloading the page, then loading the sample blink program produces the following in the javascript console:

Navigated to http://scratchx.org/?url=http://khanning.github.io/scratch-arduino-extension/arduino_extension.js
4 arduino_extension.js:508Attempting connection with COM3
arduino_extension.js:521 Uncaught TypeError: Cannot read property 'set_receive_handler' of null(anonymous function) @ arduino_extension.js:521
3 arduino_extension.js:508 Attempting connection with COM3

And, by the way, when all this happens, the chatter with the board starts again.  RX, TX,and flashes too fast to count on the special port 13 LED.

I see the gear logo on the top line of the page, right next to the three horizontal bars for settings. But nothing shows up on chrome://plugins about this plugin, even though its symbol is on the page top.

So I am at a loss.  I think that the helper function is somehow not talking to the plugin.  I have uninstalled the plugin and reinstalled it.  

I wanted to see if I could get this to work because my 8 year old grandson has said he in interested in robotics.  We are going to build a MeArm and I really would like it if he could program it.  I am not sure I can teach him C, in fact I have my doubts that I could teach anyone C, despite years of coding in it.  So if someone could help I would appreciate it.

GrandWizzardTheodore, 2016-04-21:
i have the same problem. Chrome yellow light will not go green! both drivers installed as per wedo.

Help!

khanning, 2016-04-26:
@Brasshopper @GrandWizzardTheodore there is currently a bug in the Scratch Extensions Browser Plugin that prevents the Arduino extension from working properly in the Chrome browser. In the meantime, you can use the Firefox browser along with the Scratch Extensions Browser Plugin for "other web browsers". The Arduino extension should work properly with this setup. Please let me know if you are not able to get it to work in Firefox. I will update this issue once the Chrome bug has been fixed.

Thank you!

tmancebo, 2016-07-14:
Hi,
I can't find the Scratch Extensions Browser Plugin for "other web browsers". The scratch link doesn't work:
https://scratch.mit.edu/info/ext_download/
Could you help me?

khanning, 2016-07-19:
@tmancebo the page you referenced is in the process of being restored. In the meantime, you can still get to the download links using the "Tips" window in Scratch:
https://scratch.mit.edu/projects/editor/?tip_bar=ext1

tmancebo, 2016-07-19:
It works!!!
Thank you so much!!!