khanning/scratch-arduino-extension #45: Scratchx for Arduino: Yellow Light, "Disconnected"

eprovencal, 2016-08-29:
I just sat down to start my first project using my arduino and scratchx.  I am new to both, but enjoy tinkering so I thought I would give it a try.

Problem: Scratchx is not recognizing my arduino

Here are the steps that I have taken:
(1) Installed Arduino software
(2) Tested Arduino software by making the LED on port 13 blink
(3) Installed the scratch extension - Windows (other web browsers)
(4) Open Scratchx in Mozilla & allow scratch.org to run scratch device & allow and remember scratchx.org to run plugins adobe flash and scratch device. (the light is yellow and says "disconnected" when I hover over it with my mouse)
(5) Follow the Scratchx Arduino tutorial using a breadboard, 220 ohm resistor, and an LED
(6) Only the LED in scratch's stage blinks, not the LED on my breadboard
(7) Test my breadboard set up - Adjust the program in the Arduino software to use pin 9 and run the program - it works - the LED on my breadboard blinks

Additional information:  When I am in scratchx and I unplug the arduino from the computer I get a dialog box message that says:

scratchx.org says:
Serial device error
Device: COM3
Error: The file handle supplied is not valid

What should I try in order to get scratchx to recognize the arduino?  I tried changing USB ports on the computer.

khanning, 2016-09-02:
@eprovencal the steps you listed are correct and the dialog message indicates that the plugin is in fact seeing the device. 

First, make sure that you are using the latest version of Firefox and the Flash plugin. Once you make sure everything is up to date, restart Firefox and try to connect the Arduino to ScratchX again. This time see if any messages appear in the web console. In Firefox you can get to the console by going to ```Tools``` > ```Web Developer``` > ```Web Console```. Paste anything you see in the console log here.

eprovencal, 2016-09-04:
Thank you!  Now it's working.  I updated Flash and also uploaded standardformatta on the arduino.

zhjohn925, 2019-02-08:
I installed scratch device plugin in firefox 49 on windows 10.  and tried out LED blinking example. However the Arduino was disconnected after a few seconds.  any suggestion how to fix this ?  thanks