khanning/scratch-arduino-extension #48: arduino dosent work

hellored, 2016-11-19:
I try to make the arduino extension work but on firefox witch is recomended the scratchx site dosnt work properly. i upload the extension file for other browsers and it does nothing it gives me the same window to give the extension file again. I tried it on chrome but i see no yellow or green button on the board. Please help me i need it to work

hellored, 2016-11-19:
I made a mistake i managed to make it in the site but still i see no arduino light in the blocks sector

RageBill, 2017-04-23:
You can try to download the "Firefox Extended Support Release" version [here](https://www.mozilla.org/en-US/firefox/organizations/all/). Firefox is now trying to move away from supporting these older plugins (Scratch Plugin) and that may be the reason why it is not working for you.

Logabe, 2017-09-01:
Are you using linux? i'm having a similar issue