khanning/scratch-arduino-extension #53: Scratch Extensions Browser Plugin no longer working for Firefox after updates

RageBill, 2017-03-28:
After Firefox has updated recently, the Scratch Extensions Browser Plugin no longer works on Firefox (it works previously). Any thoughts on how to solve this? Thank you!

RageBill, 2017-03-28:
Just find a workaround at the moment - by using this https://www.mozilla.org/en-US/firefox/organizations/all/ Extended Support Release version of FireFox. However, the support will cease in early 2018. Will we have any long term solution?

khanning, 2017-03-31:
@RageBill thanks for reporting this. As you discovered, Firefox recently removed support for NPAPI plugins. Between this and the fact that the Scratch device plugin for Chrome is not currently working with serial devices, the only option is to use [Firefox ESR](https://www.mozilla.org/en-US/firefox/organizations/).

The Scratch Team is aware of the issue and will hopefully have a longterm solution in the not too distant future.

closed 2017-03-31 by khanning