khanning/scratch-arduino-extension #55: Fixed connection problem with scratchx and Arduino

giulai, 2017-05-17:
Hello

In the last few days I have tried to use the extensions of scratchx and Arduino on different PCs.
The problem was always the same, it was connected for a few seconds and then disconnected.

I found the solution that seems to be stable.
- First install Firefox version 49.0.2 http://filehippo.com/download_firefox/71084/
- After you open Firefox, install the plugin available on https://create.arduino.cc
Just select "Arduino Web Editor" and create an account if you do not already have it.
You are prompted to download and install the plugin and accept the certificates.

At this point, the recommended procedure for Scratchx and Arduino works and is stable.

carlosperate, 2017-09-05:
Thanks @giulai, been experiencing the same problem, hopefully your suggestion will help!

thoreson, 2017-10-14:
thanks that was diving me nuts, glad to finally get it running.

zhjohn925, 2019-02-06:
I have problem to install the plugin on https://create.arduino.cc.  arduino icon is shown in system tray. but firefox keeps to looking for plugin forever.  I have windows 10.  firefox 49.0.2 (32-bit).  Can you see what I missed.  thanks.

Jennice16, 2019-02-06:
I solved some of my issues by downloading a firefox before version 46. They're available as offline installs, and and when installing in their own directory, then can co-exist with the more recent versions. Just create a shortcut directly to the old version. Give it a try. :)

zhjohn925, 2019-02-08:
thanks for your suggestions.  I tried out firefox 45 but had no luck with the same issue firefox is not able to find the installed plugin.