khanning/scratch-arduino-extension #65: Dosn't work on linux.

Logabe, 2017-09-01:
Hi there, i've got an arduino uno and i'm using scratch to program it. I'm using linux and i can't connect the uno to scratch. i'm doing the steps **exactly**. Hope you can help.

RageBill, 2017-09-21:
Any error messages?
What browser are you using?
There's a chance that a newer browser version is not supporting the software so it does not connect, at least that's the case for me recently. Have to switch to a compatible version.

Logabe, 2017-09-22:
1. No
2. I'm using firefox 

P.S: I tried using firefox BETA (nightly) on a windows computer and it didn't load the page **At All**.

RageBill, 2017-09-24:
I think Firefox dropped the support for certain plugins Scratch uses recently.

You can try with the [Firefox Extended Support Release](https://www.mozilla.org/en-US/firefox/organizations/all/). It worked for me.

Hope this helps :)