khanning/scratch-arduino-extension #69: Not connected to Arduino in Chrome

kissenger, 2017-11-19:
Hi,

Trying to get scratch to work with the Arduino but struggling. I have installed the extension, but get a yellow "not connected" icon on the scratchx site.  I can connect to the Arduino through the Arduino Create webpage, and have uploaded the Firmata sketch but cannot get the scratchx site to talk to the Arduino.  I get the following messages "Attempting connection with COM4".

I also tried with Firefox 49 but cannot seem to get the extension to register, as the icon stays red. Altogether not having fun - can you help?

Thanks.

KKawase0104, 2017-11-20:
Hi,

Sorry, I am trying to jump in. Because I have same problem, too. maybe.

I tried Chrome ver:62.0 and Firefox ver:57.

I learned that Scratch X use "Scratch Device Plugin" to connect arduino board by serial port, 
and "Scratch Device Plugin" use NPAPI plugins, but Chrome and Firefox is not support NPAPI plugins, no longer.
https://github.com/LLK/scratchx/issues/116

Is Scratch X not support serial port, any more ?
Do we have to make new "Scratch Device Plugin" ?

ArduinoJava, 2017-12-18:
I'm on Safari and I have the same problem!

KKawase0104, 2017-12-18:
I find s2m.
https://github.com/MrYsLab/s2m

I could connect Arduino base board to ScratchX by serial port by using s2m method.
https://mryslab.github.io/s2m/s2m/

fredbi, 2018-01-08:
@KKawase0104 good tip. Since s2m looked more suited to micro:bit, I tested this one from the same author (and same method) for Arduino; https://github.com/MrYsLab/s2aio.

It creates a python3-based bridge between scratch offline editor and an arduino board.  Worked fine for me.

I loos like all browser-plugin based solutions are not working at the moment, whatever the browser.